CSUFTitanRover/TitanRover #18: Raspi loses time accuracy

Weffe, 2017-02-24:
When the raspi is left unconnected to the internet for long time periods its unable to sync up the Global UTC. This causes its local time to become inaccurate which affects the timestamps sent in each data packet to be stored on the MongoDB Server.

Weffe, 2017-02-24:
In order to prevent this from happening just make sure to connect it to the internet and force a time sync. We should do this right before starting any missions to make sure timestamps are accurate.

shan224, 2017-02-25:
@Weffe im dont think we'll have any internet access out in the desert
@stewartjustind right?

skrapmi, 2017-02-25:
I can hook up an RTC for it to pull time from.

Weffe, 2017-02-25:
If this is the case, we either need to test how long it takes for the pi to go out of sync or find a more reliable value for timestamps which I don't know what that could be. Couldn't we also use someone's phone as a wifi hotspot to sync up time?

stewartjustind, 2017-02-25:
@weffe Problem with using someones phone is that out in the desert, I am pretty sure our cell coverage was close to nothing as well. I think @skrapmi had a good idea, just hook up an RTC that we can use to keep the timing constant.

Weffe, 2017-02-25:
Okay looks like we've solved this issue.

closed 2017-02-25 by Weffe